WeActStudio/ArduinoCore-AT32F4 #15: DFU upload stopped working on release 003.

gulweb, 2022-08-29:
The DFU upload stopped working after installing release 003. Arduino console supplied...below

- WeAct Studio Official Link
- taobao: weactstudio.taobao.com
- aliexpress: weactstudio.aliexpress.com
- blog: www.weact-tc.cn
/home/webjorn/.arduino15/packages/WeActStudio/tools/Artery_ISP_Console/0.0.1
------------------------------------------------------------
Artery ISP Programmer V3.0.01
Support USART and DFU
------------------------------------------------------------

DFU Connect......
Device:  AT32F403ACGU7    Flash:  1024 KB
PID:  70050347      BID:  4703    Version:  3.2
Device connected Successfully

/home/webjorn/.arduino15/packages/WeActStudio/tools/Artery_ISP_Console/0.0.1/DFU_download.sh: line 12:  2803 Segmentation fault      (core dumped) $(dirname $0)/AT32_ISP_Console -dfu -d --fn "$1" --v --o -r
Upload Quit.
Have a good day!

gulweb, 2022-08-29:
Note that path is 0.0.1 while release is 0.0.3

WeActStudio, 2022-08-29:
fix

closed 2022-08-29 by WeActStudio