WeActStudio/ArduinoCore-AT32F4 #18: PWM does not work for multiple pins

ShaggyDog18, 2022-09-01:
PWM is configured for 25kHz, 50% duty cycle (see the code below). Tested PWM for several pins. 
PWM works for pins: PB14, PB0, PB1, PB6, PA7, PA8
PWM does not work for: PB2, PB12, PB13, PB15, PA15, PA12 
I interested in PA15 because it is already traced at PCB.  Did not test other pins. The test code is simple:
```
#define PWM_PIN  PA15      //works for: PB14, PB0, PB1, PB6, PA7, PA8

void setup() {
  PWM_Init( PWM_PIN, 100, 25000UL );
  PWM_Write( PWM_PIN, 50 );
}
 
void loop() {
}
```

WeActStudio, 2022-09-04:
PA15 is not support

closed 2022-09-04 by WeActStudio

ShaggyDog18, 2022-09-05:
I have not find in the A32F4 datasheet any mentioning that some pins do not support PWM. May you share any document or link describing limitations?

ShaggyDog18, 2022-09-06:
regarding PA15:  the function  `IS_PWM_PIN( PA15 ) ` returns TRUE ???

ShaggyDog18, 2022-09-08:
are you going to fix it?