WeActStudio/ArduinoCore-AT32F4 #30: Availability of UART6 in HardwareSerial

ShaggyDog18, 2022-11-10:
Please, add UART6 to the HardwareSerial. 
You may also add UART7, I believe it would be required sooner or later.
Thank you!

WeActStudio, 2022-11-21:
Added, uart6_tx(pa4), uart6_rx(pa5), uart7_tx(pb4),  uart7_rx(pb3), uart8_tx(pc2),  uart8_rx(pc3)

ShaggyDog18, 2022-11-21:
Should HardwareSerial.h contain the following:
```
#if SERIAL_6_ENABLE
extern HardwareSerial Serial6;
#endif
#if SERIAL_7_ENABLE
extern HardwareSerial Serial7;
#endif
#if SERIAL_8_ENABLE
extern HardwareSerial Serial8;
#endif
```

WeActStudio, 2022-11-22:
ok

ShaggyDog18, 2022-11-23:
Thank you for a quick fix!

ShaggyDog18, 2022-12-31:
I have just started using UART6 and UART7 in my project and they does not work! 
Tested baud rates 115200, 57600 (default SERIS_8N1) and simple `Serial6.println() Serial7.println()` does not work... 
Neither `Serial.write(), Serial.read() Serial.available(`)  methods does not work.
Please, check then up. Thank you.