WeActStudio/ArduinoCore-AT32F4 #33: Incorrect output at UART with PARITY_EVEN (SERIAL_8E2) configuration

ShaggyDog18, 2022-12-21:
I experience an incorrect data output at USART3, though it might be at all UARTs.  
The output: sequence of bytes at baud rate 115200, SERIAL_8E2 (also tried a custom 100000: the baud rate does not influence on the issue).  The output bytes are not correct. However, the UART works fine for Parity NONE configuration (SERIAL_8N2).
Transfer data by `serial.write( uint8_t *byte, nBytes )` method. Also tried `serial.write( uint8_t byte)` method with the same issue.
Here is the sequence of bytes: the line L1 is the output sequence; the line L2 is the actual output caught at the TX. I marked by * the bytes that are not matching (at the end of the incorrect byte pairs) . 
```
L1: 00001111  11101010*  01010011  10011111  11111010  11010100  10100111*  00111110*  10001011  01011000*  01000100  01111101  10110011*  10110110*  10001000  01000101*  11010100  00000111* 
L2: 00001111  01101010*  01010011  10011111  11111010  11010100  00100111*  10111110*  10001011  11011000*  01000100  01111101  00110011*  00110110*  10001000  11000101*  11010100  10000111*
``` 
As you can see, the difference has appeared in the first leading bit of transferred bytes, sometimes bytes are correct (no *), sometimes '0' is swapped by '1' and wise versa. I have not spotted the regularity.

Please, take a look at how EVEN parity is processed during data transfer (TX) and fix.
Thank you!
P.S.  I spent almost 3 weeks identifying the possible cause of the incorrect data output! Please, need your help to fix. Thank you.

WeActStudio, 2022-12-25:
fixed

closed 2022-12-25 by WeActStudio

ShaggyDog18, 2022-12-25:
Thankyou!
Why you've deleted definitions of the UARTs 6,8, and 8 from the `HardwareSerial.h` :
```
#if SERIAL_6_ENABLE
extern HardwareSerial Serial6;
#endif
#if SERIAL_7_ENABLE
extern HardwareSerial Serial7;
#endif
#if SERIAL_8_ENABLE
extern HardwareSerial Serial8;
#endif
```

ShaggyDog18, 2022-12-25:
I am testing and I am confused: SERIAL_8E2 works fine at 115200 baud rate. However, as soon as I switch to a required custom baud rate of 100,000 - the data are distorted. The baud rate did not influence on the output before, now it does...
My project works with custom UART configs of 100.000 baud rate and SERIAL_8E2, so it is essential to get the output right at custom UART configuration.  Please, take another look. Thank you!

ShaggyDog18, 2022-12-25:
Works now at my custom config for 100.000 baud rate and 8E2 ! 
Thank you  for your prompt support!!!