WeActStudio/ArduinoCore-AT32F4 #43: Support request: Serial upload.

XanCraft21, 2023-04-19:
Hello, i was interested in this boards package, but there is no serial upload support. I can not program this board without being able to use a FTDI adapter as the usb boot loader your company uses seems to never give a valid port on Windows regardless of if i have a driver or not. If you could be able to add this support i will gladly use the board. Thank you for your help.

ShaggyDog18, 2024-03-03:
you may try uploading by USB (use USBC cable, switch the board to a DFU mode by pressing BOOT0 key while connecting to USB, or reset while press & holding BOOT0 key right before uploading .bin file). It works smoothly.

XanCraft21, 2024-03-04:
Unfortunately DFU mode has always given me problems. It could also be because the software is not too well supported for pretty much every device using DFU for programming in general.

I can never get it to work since all the available software is sketchy and kind of limited. Because of this i can only use an FTDI adapter.

Possibly if an HID bootloader was used instead it would work for me. Until then i cannot use the board without Serial upload support. But thank you for the tips though.

XanCraft21, 2024-06-22:
I have figured it out, i managed to make the uploading work in Serial mode. I had to modify the WeAct Studio Artery Programmer batch file under appdata, local, arduino, packages within the board files. The picture below shows the flags i used, note that the port must be set manually. To edit the file you need to change the extension to txt.
[image]

closed 2024-06-22 by XanCraft21