WeActStudio/ArduinoCore-AT32F4 #45: Cannot install from board manager

s1nisteR, 2023-10-29:
I added the board URL but nothing shows up in my board manager. This library is unusable from Arduino IDE Version 2.

ShaggyDog18, 2024-03-03:
strange, it works for me smoothly...

LUSTgabriel, 2024-04-15:
I cant use it either....here is my env: Arduino ide version2.3.2 cli version 0.35.3 cpu 7950x windows 10 22h2, i dont think so is my computer promblem, because other lib can be installed correctly.

LUSTgabriel, 2024-04-15:
I test the arduino 1.8.18, it works well, plese figure out why this lib can not work on arduino ide version 2.x.x uper

LUSTgabriel, 2024-04-15:
I have a intresting descovered...... since i installed arduino 2.3.2 and 1.18.3 same time, and also same place, when i install json file on the old version it seems sync to the new verison well on the application .......why?????
[image]

jlsilicon, 2024-05-22:
Useless, I cant load it either, not JSON nor ZIP file - arduino (1.8.19 & 2.3.2) just says bad zip not valid library.

-- Do you intyend to fix this prob someday ... ?

KenjutsuGH, 2024-05-30:
> I have a intresting descovered...... since i installed arduino 2.3.2 and 1.18.3 same time, and also same place, when i install json file on the old version it seems sync to the new verison well on the application .......why????? !

I am seeing this with the macOS versions as well. Version 1.8.19 will install the board manager files but not Version 2.3.2. After installing with 1.8.19, 2.3.2 will pick it up and can use it.

koendv, 2024-11-30:
It's a security problem. On 2.3 the log says "Checking signature" and then "Missing signature file".  Looking for a file "package_at32_index.json.sig"?

ggatis, 2025-03-24:
Please make it available for installation on Arduino 2.3.4.