WeActStudio/ArduinoCore-AT32F4 #47: WeActStudio support is very inadequate, almost non-existent

rizacelik, 2024-01-14:
WeActStudio produces great cards, but it's far from user-friendly help that I can see. Ignoring the questions and problems here will earn you minus points.
ArduinoCore-AT32F4 library works well with Arduino. After the code is loaded, there is no support for sending it to the screen from the Serial interface. Serial COM PORT support is missing or we don't know. We are waiting for an explanation on this matter.

ShaggyDog18, 2024-03-03:
Serial is supported for sure: there are 6 serial ports! 
you need to include Hardware Serial where Serials are defined:
`#include <HardwareSerial.h>`
Then use them as Serial, Serial2, Serial3, ... Serial6.
For TX-RX pins refer to the AT32 chip datasheet.

rizacelik, 2024-03-03:
Thank you very much for your reply. Where can I download version 0.0.8? Here we have version 0.0.1.

ShaggyDog18, 2024-03-04:
Current files in the repository is v0.0.8.

rizacelik, 2024-03-05:
v0.0.8 installed. USB COM PORT is not active. So I can't use Serial Monitor.
I'm waiting for your help on this matter. What settings should I make?

[image]

Example code:

`#include <HardwareSerial.h>

HardwareSerial Serial1(USART1);

// the setup function runs once when you press reset or power the board
void setup() {
  Serial1.begin(9600);
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED_BUILTIN, HIGH);  // turn the LED on (HIGH is the voltage level)
  Serial1.println("LED");
  delay(500);                      // wait for a second
  digitalWrite(LED_BUILTIN, LOW);   // turn the LED off by making the voltage LOW
  delay(500);                      // wait for a second
}
`

ShaggyDog18, 2024-03-05:
1. you do not need the second line. all  UARTS are already mapped in `HardwareSerial.h `
USART1 is pre-assigned to `Serial`. so you refer to it as, for example:
```
void setup() {
  Serial.begin(9600);
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}
```
3. Serial is not mapped to USB.  UART1 is on pins TX PA9-RX PA10 (similar to STM32). 
So, you have to connect UART-USB adaptor to pins in order to communicate with your PC.

rizacelik, 2024-03-05:
Thank you. I will try.